BroadwellConsultingInc/SerialWombat #17: includes missin capitable letters

bm16ton, 2021-08-13:
So on every install i have to manually edit the includes IE
#include "serialWombat.h"
because its missing the correct capital letters, i finally realized your probly using windows, or some other god awful OS and wouldn't notice. I love your product! But I really really wish it was more open source friendly. It would be great to be able to compile its firmware with opensource toolchains. Even the sniffer addon is for closed source tools. Any chance the future 20pin version will be more FOSS friendly? Its really the only downfall of these awesome little guys! Thank you for your awesome work/product! I may have a suggestion, the upcoming raspberrypi libs...They gonna be python? The raspberrypi landscape is littered with old deprecated code for cool things like this based on python etc, and from userland. Kernel module drivers for something like this is in the long run a much better, choice that will hold up much more easily and be able to integrate into the landscape more, make the interfacing with it as simple as cat and echo....and its the correct way :) lol. When I have time Im hopeful ill get around to writing at least sum modules for its various functions. I realize its probly a falling on deaf ears situation but its worth a good think on.

BroadwellConsultingInc, 2021-08-14:
Thanks for the bug report.  You are correct that I develop on Windows.  I'll get an update to the Arduino library pushed out soon with a fix for the capitalization .  I'll create an Ubuntu VM and make sure the example projects compile under Arduino in that environment.

Glad you like the product.  I tried hard to really anticipate needs and understand areas where Arduino is weak and try and supplement those areas.   

I love open source code.  But ultimately to be a great product the code has to run on a Micro that's well suited for the I/O operations the Serial Wombat firmware delivers.  The PIC16F15214 was far and away the best choice currently available in terms of breadboard-friendly package, peripheral features, supply voltage range, cost and available flash space for pin modes, so that's what I used.  It was a hard choice to decide to use the subscription-based Pro version of the XC8 compiler for this, but that was the path to create the very highest performance device possible (short of coding in assembly, which probably would have lead to a first release sometime in 2022 or 2023).

The Serial Wombat 18B 28 Pin version (18 available IO Pins)  will similarly use the PIC24FJ256GA702 and XC16/MPLAB chain for the same reasons.  The Serial Wombat 4B code is designed to be portable with abstracted hardware calls in the pin modes so that porting the concept and high level protocol and pin mode code to other microcontrollers should be straightforward if someone wants to port to a platform that's more open source.  The upcoming SW18B's firmware design is very tightly tied/optimized to the 24FJ256GA702's DMA system and peripherals, so it will be less portable (but should really be awesome).

I've started some work on a Raspberry Pi Python lib but it's slow going, as my primary focus is Arduino and I'm not very experienced with Python.   I don't anticipate creating Kernel module drivers as this is outside my are of expertise, and I think my development time is probably best suited staying focused on the Serial Wombat firmware.  If you want to take a crack at it, I'd be happy to support you in any way I can and would love to feature your work in the intro to one of my upcoming YouTube videos if you get something working.

BroadwellConsultingInc, 2021-08-14:
Actually, this is a bug for the Arduino Library, so I'm closing this one out and opening a new bug in that repository.

closed 2021-08-14 by BroadwellConsultingInc

bm16ton, 2021-08-14:
You are awesome and I totally get it! I'll try and prioritize the kernel
drivers, I guess with all its functions itll be an end which I've never
done before, should be a alotta fun and learning. I'll start with the adc
as those are fun to play with! I love the concept of this product so much.
Very happy and a like proud to be one of the people who bought from your
first batch. I was using appollo3, teensy, and stm32 in arduino and was
having problems anytime I started the adc example wo anything attached to
the adc pins it would either give bad readings or more likely no readings,
only the apollo3 could be fixed by unplugging and unplugging the wombat
with mcu staying powered the others wouldn't recover. No biggy a couple
resistors fixed that but then life got in the way. I've never submitted to
the linux kernel before and am well a crap wanna be programmer but I've
eaked out a dozen or more kernel drivers. I just know I'll be able to use
these devices years from now no problems if I can get it mainline and of
course its features and available tools match up so it immediately brings
working apps on the table. I really hope this concept takes off for you and
am excited to see what you make next! I'll def be on the first batch of the
20pins if it's possible. Well thought out product my friend, makes things
easier. thank you