BroadwellConsultingInc/SerialWombat #18: Fail to compile on SAMD51

LaudixGit, 2021-08-31:
Can compile and load Blink sketch.
Cannot compile any SerialWombat examples.

compiler output:

`Arduino: 1.8.15 (Linux), Board: "Adafruit Feather M4 Express (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

/opt/arduino-1.8.15/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.15/hardware -hardware /home/laudix/.arduino15/packages -tools /opt/arduino-1.8.15/tools-builder -tools /opt/arduino-1.8.15/hardware/tools/avr -tools /home/laudix/.arduino15/packages -built-in-libraries /opt/arduino-1.8.15/libraries -libraries /home/laudix/code/libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_8022 -ide-version=10815 -build-path /tmp/arduino_build_620646 -warnings=none -build-cache /tmp/arduino_cache_465779 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.bossac.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2 -prefs=runtime.tools.CMSIS-Atmel-1.2.2.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2 -prefs=runtime.tools.arduinoOTA.path=/home/laudix/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/laudix/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=/home/laudix/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/laudix/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -verbose /home/laudix/code/libraries/SerialWombat/examples/SW4B_Ard_WombatFinder/SW4B_Ard_WombatFinder.ino
/opt/arduino-1.8.15/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.15/hardware -hardware /home/laudix/.arduino15/packages -tools /opt/arduino-1.8.15/tools-builder -tools /opt/arduino-1.8.15/hardware/tools/avr -tools /home/laudix/.arduino15/packages -built-in-libraries /opt/arduino-1.8.15/libraries -libraries /home/laudix/code/libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_8022 -ide-version=10815 -build-path /tmp/arduino_build_620646 -warnings=none -build-cache /tmp/arduino_cache_465779 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.bossac.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2 -prefs=runtime.tools.CMSIS-Atmel-1.2.2.path=/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2 -prefs=runtime.tools.arduinoOTA.path=/home/laudix/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/laudix/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/laudix/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=/home/laudix/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/laudix/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -verbose /home/laudix/code/libraries/SerialWombat/examples/SW4B_Ard_WombatFinder/SW4B_Ard_WombatFinder.ino
Using board 'adafruit_feather_m4' from platform in folder: /home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4
Using core 'arduino' from platform in folder: /home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4
Detecting libraries used...
/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"SW4B_Ard_WombatFinder.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10815 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/libraries/Adafruit_TinyUSB_Arduino/src/arduino -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/libraries/Adafruit_TinyUSB_Arduino/src/arduino -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2/CMSIS/Device/ATMEL/ -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/cores/arduino -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/variants/feather_m4 /tmp/arduino_build_620646/sketch/SW4B_Ard_WombatFinder.ino.cpp -o /dev/null
Alternatives for SerialWombat.h: [SerialWombat@1.0.9]
ResolveLibrary(SerialWombat.h)
  -> candidates: [SerialWombat@1.0.9]
/home/laudix/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"SW4B_Ard_WombatFinder.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10815 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/libraries/Adafruit_TinyUSB_Arduino/src/arduino -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/libraries/Adafruit_TinyUSB_Arduino/src/arduino -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/home/laudix/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.2/CMSIS/Device/ATMEL/ -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/cores/arduino -I/home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/variants/feather_m4 -I/home/laudix/code/libraries/SerialWombat/src /tmp/arduino_build_620646/ske [... cut ...]
Alternatives for serialWombat.h: []/home/laudix/code/libraries/SerialWombat/src/SerialWombat.cpp:1:10: fatal error: serialWombat.h: No such file or directory

 #include "serialWombat.h"ResolveLibrary(serialWombat.h)

          ^~~~~~~~~~~~~~~~
  -> candidates: []compilation terminated.

Using library SerialWombat at version 1.0.9 in folder: /home/laudix/code/libraries/SerialWombat 
Using library Wire at version 1.0 in folder: /home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.4/libraries/Wire 
exit status 1
Error compiling for board Adafruit Feather M4 Express (SAMD51).
`

BroadwellConsultingInc, 2021-08-31:
Thanks for submitting this.  I'm releasing a fix for this today.  This is a bug caused by my being careless with case on include files.  I develop on Windows so it doesn't show up but it will on case-sensitive file systems.

I created a copy of this bug in the repo for the Arduino Library so I can link the release to it:
https://github.com/BroadwellConsultingInc/SerialWombatArdLib/issues/1

closed 2021-08-31 by BroadwellConsultingInc

BroadwellConsultingInc, 2021-08-31:
Fixed in V1.0.10 of the Arduino library.  Pull in the latest version from the library manager.  Tested on Ubuntu in an VM, with Adafruit D51 board package and Feather M4 board as target.  All examples compiled.

LaudixGit, 2021-08-31:
Thank you! 
I can confirm the WombatFinder example compiles, uploads, and runs as expected.

`
Serial Wombat Finder

I2C Device found at address 0x6E
Serial Wombat Found.
Model: 04B
FW Version: 203
Source voltage (mV):  3350
`