BroadwellConsultingInc/SerialWombat #19: Ultrasonic sensor not returning valid data

ctacke, 2022-07-22:
I'm trying to get the ultrasonic data working on an 18AB, but the data I'm getting back isn't valid (it just sits at 0 or 1). I'm using an HC-SR04, and writing in C#.  There are no complete samples in the Arduino, Python or C# libraries that I can find.  I'm using I2C to communicate with the 18AB and the data looks like this:
```
// Configure to use WP10 and WP11
TX C8-0A-1B-00-0B-00-01-55
RX C8-0A-1B-00-0B-00-01-55

// read public data
TX 81-0A-FF-FF-55-55-55-55
RX 81-0A-00-00-55-55-55-55
TX 81-0A-FF-FF-55-55-55-55
RX 81-0A-00-00-55-55-55-55
```
If I read the pulses, I get a number that only appears to increment periodically - like every 5+ seconds or so, not a continually increasing number.

Do you have source for a sample that just configures and then reads the distance?

ctacke, 2022-07-23:
I'm using 5V, and doing nothing else at all with the device, so no other commands are occurring.

I was only running 3.3 - I changed it to 5V and now I get data, but the behavior on the scope is odd.  Sometimes the `trig` pulses are 1ms apart and the results then seem about right, but, sometimes they are 143-ish ms apart and then the distance values are way off - like reading 140cm when it should be more like 8.

Since I have only 1, I'm going to order another sensor just to rule that out.

BroadwellConsultingInc, 2022-12-19:
Closing as unable to reproduce.

closed 2022-12-19 by BroadwellConsultingInc