BroadwellConsultingInc/SerialWombat #20: Can't build firmware for the 18A

gtjoseph, 2022-07-27:
Is there a guide somewhere for actually building the firmware?   I have mplab_ide 6.0 and the xc16 4.5.1 installed fine but the code is looking for the Arduino headers (via SerialWombatArdLib:SerialWombat.h).   After adding the ArduinoCore-API and ArduinoCore-avr repos to the project's include paths it still doesn't work because the xc16 compiler doesn't understand c++.   So what are the steps that have to be done to get a working build?

Thanks!

BroadwellConsultingInc, 2022-07-27:
I think it's probably a path setup issue.  The SW18AB refers to a lot of header files in the SerialWombat:SerialWombatCommon directory that provide some common constants between the SW18AB and SW4B firmware.  
https://github.com/BroadwellConsultingInc/SerialWombat/tree/main/SerialWombatCommon

The SW18AB build needs to have two parts:   The bootloader or bootloader jump to app set up as the main project, and then the SW18AB application.  I'll put together a quick and dirty walkthrough video this weekend on building the firmware.  I'm currently using MPLAB 5.45 and XC16 V1.7.0

gtjoseph, 2022-07-27:
That'd be great, thanks!

FYI... Here's an example of the issues I'm getting now...
```
In file included from /usr/src/mcu/PIC/Wombat/SerialWombatArdLib/src/SerialWombat.h:31:0,
                 from pic24fj256ga702/mcc_generated_files/i2c2.c:49:
/usr/src/mcu/arduino/ArduinoCore-API/api/HardwareSerial.h:24:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arduino'
In file included from /usr/src/mcu/PIC/Wombat/SerialWombatArdLib/src/SerialWombat.h:32:0,
                 from pic24fj256ga702/mcc_generated_files/i2c2.c:49:
/usr/src/mcu/arduino/ArduinoCore-avr/libraries/Wire/src/Wire.h:34:7: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'TwoWire'```
It looks like the compiler doesn't like the "namespace" or "class" keywords in the arduino header files..

There are also a whole bunch of files (nbproject and .generated_files) that get modified as soon as you open mplab on a clean repo checkout.  It makes things a little confusing.  You might want to add them to a .gitignore file.  You might also want to remove the .hex files from the releases directory and just attach them as assets to the github releases.   This way the repo doesn't get polluted with non-code changes.

BroadwellConsultingInc, 2022-07-27:
Thanks.  I'm still on the Git learning curve.  I'll work on the cleanup soon.  Looking at your captures, it occurs to me that you might be having filename case sensitivity issues.  I develop on Windows where such things don't cause a lot of problems, so I am sometimes careless in that regards.

In any case, you definitely don't want to include any files from the Arduino repository.  Everything you need should be in the SerialWombat18A_18B and SerialWombatCommon directories of the repository.

I took a look at Microchip's site and it appears that MPLAB 6 doesn't officially support the PicKit 3.   I use a PicKit 4 because it's much faster, but I want to stay with an IDE that supports the PicKit 3 as long as possible because it's a much cheaper way for users to get started, and it's supported for the PIC24FJ256GA702 in MPLAB 5.45 .   So I don't anticipate offically supporting later releases of MPLAB until there's no choice.

BroadwellConsultingInc, 2022-07-27:
I installed MPLAB X 5.45 and X16 V1.70 in an Ubuntu VM.  I fixed the case issues on the include files, and it built.  See if you can do the same.

gtjoseph, 2022-07-28:
Ah gotcha. I don't even have a Windows machine anymore so I forget about the case insensitivity thing.  So it was always supposed to be serialWombat.h not SerialWombat.h.  I'll give it a try in the morning.

gtjoseph, 2022-07-28:
Success!  With the latest commits I was able to build with both 5.45/1.70 and 6.00/2.00.  The only issue was a warning in both environments...
```
Info: Loading file: ./p24FJ256GA702_boot.gld
"/opt/microchip/xc16/v1.70/bin/xc16-gcc"   -o dist/default/production/SerialWombat18A_18B_Bootloader.X.production.elf  build/default/production/_ext/275911920/adc1.o build/default/production/_ext/275911920/clock.o build/default/production/_ext/275911920/crc.o build/default/production/_ext/275911920/cvr.o build/default/production/_ext/275911920/dma.o build/default/production/_ext/275911920/i2c1.o build/default/production/_ext/275911920/i2c2.o build/default/production/_ext/275911920/interrupt_manager.o build/default/production/_ext/275911920/mcc.o build/default/production/_ext/275911920/pin_manager.o build/default/production/_ext/275911920/system.o build/default/production/_ext/275911920/tmr1.o build/default/production/_ext/275911920/tmr2.o build/default/production/_ext/275911920/tmr3.o build/default/production/_ext/275911920/traps.o build/default/production/_ext/275911920/uart1.o build/default/production/_ext/275911920/uart2.o build/default/production/_ext/1542527744/flash.o build/default/production/main_boot.o build/default/production/protocol_boot.o build/default/production/_ext/1497480854/pinDigitalHwSpecific.o build/default/production/_ext/1497480854/wombatasm.o build/default/production/_ext/998979777/asciiConversion.o      -mcpu=24FJ256GA702        -omf=elf -DBOOT -DXPRJ_default=default  -legacy-libc    -I"../SerialWombat18A_18B.X/pic24fj256ga702/mcc_generated_files" -I"../../SerialWombatCommon" -I"../SerialWombat18A_18B.X/pic24fj256ga702" -Wl,--local-stack,,--defsym=__MPLAB_BUILD=1,,--script="p24FJ256GA702_boot.gld",--stack=16,--check-sections,--data-init,--pack-data,--handles,--isr,--no-gc-sections,--fill-upper=0,--stackguard=16,--no-force-link,--smart-io,-Map="dist/default/production/SerialWombat18A_18B_Bootloader.X.production.map",--report-mem,--memorysummary,dist/default/production/memoryfile.xml  -mdfp="/home/gtjoseph/.mchp_packs/Microchip/PIC24F-GA-GB_DFP/1.4.141/xc16" 
/opt/microchip/xc16/v1.70/bin/bin/../bin/elf-ld: Warning: Invalid FBSLIM setting.
  calculated end address of 0x004800   is not within bounds:
  0x000100   to 0x003ffe   for 'program' region
/opt/microchip/xc16/v1.70/bin/bin/../bin/elf-ld: Warning: program memory region not defined, using default range: 0x100 .. 0x3fffe
```

So now that I've got it build, does SW18ABPostProcessing and/or HexToLines need to be run?  If so, what are the command line arguments to each?

I still have an issue flashing in both environments but that may be my pickit 3 acting up.  I have a pickit 4 coming in later today.

BroadwellConsultingInc, 2022-07-28:
Great.  That warning is expected.  The FBSLIM config register has to do with where the Alternate vector table gets placed.  There seems to be some "Helpful" features in the compiler that will work against settings you make vs. the part size vs. what's in the linker script, instead of just leaving you alone and setting it like you command.  I've not found a way to eliminate this warning.

closed 2022-07-28 by BroadwellConsultingInc

gtjoseph, 2022-07-28:
> So now that I've got it build, does SW18ABPostProcessing and/or HexToLines need to be run? If so, what are the command line arguments to each?

Oops, before you close... what about the post processing stuff?

BroadwellConsultingInc, 2022-07-28:
No.  That's some utilities I wrote that calculate and insert a CRC, and then generate a run-length-encoded C array of the application area.  These are used to create the Arduino firmware update sketches.   When you have your PK4 working you should be able to just hit build and download, and the firmware should be ready to go, listening on whichever interface is ready.  Note that you should be buildign and setting as your main project either the bootloader or the jump to boot app.  Both include the main application as a sub project.   I use the bootloader if I'm programming for production, and the jump to boot if I'm debugging, because there are conflicting symbols in the bootloader and application that confuse the debugger.

gtjoseph, 2022-07-28:
Perfect, thanks for your help!

gtjoseph, 2022-07-29:
Just FYI...   I was able to finally build and flash successfully.  Some things I discovered that may be of use to others, especially if you're going to do a video on building and programming or update the README.

* The ZIF adapter board that came with my PICKIT 3.5 was not wired correctly for the PIC24.  
  * The PGD and PGC pins on the board were wired to pins 15 and 16 on the PIC which are Vdd/AVdd and Vss/AVss respectively (bad) 
  * The MCLR line didn't have the recommended 10K pull-up.
  * The Vcap PIC pin wasn't connected to Vss with the recommended 10uf cap.
* If you get the invalid devi8ce id 0xffff0000 message, it's probably the above issue.
* Both the PICKIT 3.5 and PICKIT 4 work fine with MPLAB X v6.00 with the PICKIT providing power to the target or supplying power externally.
* MPLAB X 5.25 has some issues under Linux (at least for me) which is why I'm using 6.00.
* You don't need to set the PICKIT on the SerialWombat18A_18B project.  Only the Bootloader or JumpToApp projects.
* Builds with the xc16 v2.00 compiler and the PIC24F-GA-GB_DFP,1.7.214 DFP seem to complete successfully although I haven't actually tried one yet.
* MPLAB IPE works fine with the PICKITS but on Linux, the settings dialog boxes don't render correctly so you can't set the programmer to power the target, you have to use external power.  MPLAB IDE doesn't have the issue.

I'm also exploring ways to isolate the NetBeans/MPLAB config files so the base config can be checked in but things like changing the programmer settings stay local.

BroadwellConsultingInc, 2022-08-01:
The video promised above is now up:

https://youtu.be/PNPlAaIrR1o

It's about an hour of explanation of the firmware architecture and implementation.  I'd suggest watching at 1.25 speed, as I am not a particularly fast talker.