BroadwellConsultingInc/SerialWombat #21: Quad encoder read with replacement not working on SW18

wilkij1, 2022-10-23:
I'm using the SW18 to read some high quality encoders connected to an NodeMCU board. The code works fine when I use the regular sw.read() to read the position but not when I use read with replacement. When I use the read with replacement the returned value is always equal to the replacement value. This is true even if I write a value to the wombat and then do a read with replacement. In that case I get back the replacement value, rather than the value that I just wrote.

I also have some some W4 chips so I tried one of them to reduce the likelihood that I'm doing something wrong. The SW4 works exactly as expected. The only modifications to the attached code are the address at initialization and the pin selections. The code and its output are attached. There are 2 constants at the top to select the SW4 vs SW18, and to use regular read or read/replace. The outputs are in the PDF.

I have the pullups disabled and 0 set for the debounce time. I've checked the signals on a scope and they are pristine, without any debounce requirement (screen capture attached).
[read_replace.zip](https://github.com/BroadwellConsultingInc/SerialWombat/files/9847482/read_replace.zip)

BroadwellConsultingInc, 2022-10-23:
Thank you for the detailed description of the problem.  This bug report is exemplary.  The SW4B / SW18AB comparison is really useful, and definitely show's there's a problem.  I'll investigate this and fix it in the next release, probably sometime around Christmas.

BroadwellConsultingInc, 2022-12-19:
Fixed in SW18AB_V208 .

closed 2022-12-19 by BroadwellConsultingInc